NVIDIA-AI-IOT/trt_pose #2: module 'trt_pose.plugins' has no attribute 'find_peaks'

hahalidaxin, 2019-10-24:
Hello!
After setup in PC, I tried to running demo, but I got :
AttributeError: module 'trt_pose.plugins' has no attribute 'find_peaks'
It seems like a error missing package under **plugins** caused by setup.py script . Do you kown how to fix it ?
THX

jaybdub, 2019-10-24:
Hi,

Thanks for reaching out!

Do you mind sharing the output of running the setup.py script?  It seems like the C++ components may not have built successfully.

Best,
John

hahalidaxin, 2019-10-25:
Output details are as bellows:

running install
running bdist_egg
running egg_info
writing trt_pose.egg-info/PKG-INFO
writing dependency_links to trt_pose.egg-info/dependency_links.txt
writing top-level names to trt_pose.egg-info/top_level.txt
reading manifest file 'trt_pose.egg-info/SOURCES.txt'
writing manifest file 'trt_pose.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/trt_pose
copying trt_pose/parse_objects.py -> build/lib.linux-x86_64-3.6/trt_pose
copying trt_pose/coco.py -> build/lib.linux-x86_64-3.6/trt_pose
copying trt_pose/draw_objects.py -> build/lib.linux-x86_64-3.6/trt_pose
copying trt_pose/__init__.py -> build/lib.linux-x86_64-3.6/trt_pose
copying trt_pose/train.py -> build/lib.linux-x86_64-3.6/trt_pose
creating build/lib.linux-x86_64-3.6/util
copying util/cam.py -> build/lib.linux-x86_64-3.6/util
copying util/__init__.py -> build/lib.linux-x86_64-3.6/util
creating build/lib.linux-x86_64-3.6/trt_pose/models
copying trt_pose/models/mnasnet.py -> build/lib.linux-x86_64-3.6/trt_pose/models
copying trt_pose/models/dla.py -> build/lib.linux-x86_64-3.6/trt_pose/models
copying trt_pose/models/resnet.py -> build/lib.linux-x86_64-3.6/trt_pose/models
copying trt_pose/models/common.py -> build/lib.linux-x86_64-3.6/trt_pose/models
copying trt_pose/models/__init__.py -> build/lib.linux-x86_64-3.6/trt_pose/models
copying trt_pose/models/densenet.py -> build/lib.linux-x86_64-3.6/trt_pose/models
running build_ext
building 'plugins' extension
creating build/temp.linux-x86_64-3.6
creating build/temp.linux-x86_64-3.6/trt_pose
creating build/temp.linux-x86_64-3.6/trt_pose/plugins
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/TH -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/THC -I/root/anaconda3/envs/pytorch1.0/include/python3.6m -c trt_pose/plugins/find_peaks.cpp -o build/temp.linux-x86_64-3.6/trt_pose/plugins/find_peaks.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=plugins -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/TH -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/THC -I/root/anaconda3/envs/pytorch1.0/include/python3.6m -c trt_pose/plugins/paf_score_graph.cpp -o build/temp.linux-x86_64-3.6/trt_pose/plugins/paf_score_graph.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=plugins -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
trt_pose/plugins/paf_score_graph.cpp: In function ‘void paf_score_graph_out(at::Tensor, at::Tensor, at::Tensor, at::Tensor, at::Tensor, int)’:
trt_pose/plugins/paf_score_graph.cpp:59:27: warning: unused variable ‘progress’ [-Wunused-variable]
                     float progress = 0.0;
                           ^
trt_pose/plugins/paf_score_graph.cpp:10:9: warning: unused variable ‘M’ [-Wunused-variable]
     int M = peaks.size(2);
         ^
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/TH -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/THC -I/root/anaconda3/envs/pytorch1.0/include/python3.6m -c trt_pose/plugins/refine_peaks.cpp -o build/temp.linux-x86_64-3.6/trt_pose/plugins/refine_peaks.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=plugins -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/TH -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include/THC -I/root/anaconda3/envs/pytorch1.0/include/python3.6m -c trt_pose/plugins/plugins.cpp -o build/temp.linux-x86_64-3.6/trt_pose/plugins/plugins.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=plugins -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/include -I/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/torch/inclu [... cut ...]
removing '/root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/trt_pose-0.0.0-py3.6-linux-x86_64.egg' (and everything under it)
creating /root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/trt_pose-0.0.0-py3.6-linux-x86_64.egg
Extracting trt_pose-0.0.0-py3.6-linux-x86_64.egg to /root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages
trt-pose 0.0.0 is already the active version in easy-install.pth

Installed /root/anaconda3/envs/pytorch1.0/lib/python3.6/site-packages/trt_pose-0.0.0-py3.6-linux-x86_64.egg
Processing dependencies for trt-pose==0.0.0
Finished processing dependencies for trt-pose==0.0.0

princefr, 2019-11-11:
Hi @hahalidaxin after building, please put plugins.cpython-36m-x86_64-linux-gnu.so who is under buid/lib.linux-x86_64-3.6/trt_pose/ into  /trt_pose/. should be fine after that.

hahalidaxin, 2019-11-11:
Appreciate your advice @princefr , I dynamically load plugins.cpython-36m-x86_64-linux-gnu.so after building and that works.

princefr, 2019-11-11:
Perfect. your welcome