NVIDIA-AI-IOT/trt_pose #5: some question about Tensorrt with pytorch

SherlockHua1995, 2019-11-15:
Hello,jaybdub.
Thanks for you code.
I am trying to accelerate the trained Pytorch model inference process with Tensorrt
but got nothing useful from searched blogs.
I found you have done this work .
Could you plz give me some instruction about this ?

jaybdub, 2019-11-15:
Hi SherlockHua1995,

Thanks for reaching out!

In this project we use torch2trt

https://github.com/NVIDIA-AI-IOT/torch2trt

There are some examples in the README.  It works for many CNN architectures.  

I maintain that project as well, so please let me know if you have any questions.

Best,
John

SherlockHua1995, 2019-11-15:
Thank for your response.
I have tried the torch2trt you mentioned before, but got some confusing error.
would you plz show me more detail about how you convert the trained pytorch model into a trt one.
If convenient , the converter script can helps me to check my code.

jaybdub, 2019-11-15:
Could you share the code you used when you attempted to use torch2trt?

It will also help to see the exact error you faced.

Best,
John

SherlockHua1995, 2019-11-15:
import torch
from torch2trt import torch2trt
from model import LSCCNN

#MODEL PATH
checkpoint_path = './weights/weights.pth'

# create some regular pytorch model...
model = LSCCNN(checkpoint_path=checkpoint_path).eval().cuda()

# create example data
x = torch.ones((1, 3, 224, 224)).cuda()

# convert to TensorRT feeding sample data as input
model_trt = torch2trt(model, [x])

y = model(x)
y_trt = model_trt(x)

# check the output against PyTorch
print(torch.max(torch.abs(y - y_trt)))

#save the trt model
torch.save(model_trt.state_dict(), 'LSCCNN_trt.pth')

SherlockHua1995, 2019-11-15:
Any advice?Thanks.

jaybdub, 2019-11-16:
Hi SherlockHua1995,

Are you able to share the model LSCCNN?  It's possible there is a layer that is not supported.

Best,
John