NVIDIA-AI-IOT/trt_pose #14: trt_pose in C++ program

SLyra21, 2019-12-05:
Hi,

I am more or less new in the field of TensorRT optimization and I would like to ask about the possibility to use the trt_pose in C++. Can I export the model to ONNX format and load it afterwards, so that TensorRT can provide the optimization for the process? Can I use the ONNX model in a C++ program to get a pose estimation?

Best regards,
Simon

thancaocuong, 2020-11-11:
@SLyra21  you can use python to export plan file from torch2trt generated .trt file. Then load it using tensorRT c++ API and inference.

batselem, 2021-04-16:
I just finished implementing it in windows C++. 
https://github.com/batselem/Human_pose_detection

thancaocuong, 2021-04-16:
@batselem  I've just take a look at your repo. But please include build system and instruction to build and run your source code. Thanks

batselem, 2021-04-16:
I will update its readme soon.

batselem, 2021-04-19:
@thancaocuong added the detailed installation guide.