NVIDIA-AI-IOT/trt_pose #17: I want to use more network models.

raspberry-pi-maker, 2019-12-15:
I recently used OpenPose, Tensorflow, etc. for pose estimation on the Jetson Nano, but did not get satisfactory performance.
But your works are best for Jetson series pose estimation! Thanks a lot.

There are only 2 models available.
I'm going to use Jetson TX2. I think that the result will be  better than Nano.
I think that the TX2 can process more heavy models.
So I want you provide more models like ResNet-50, ...

IsmaelElHouas, 2020-05-06:
Hi @raspberry-pi-maker 

I m also interested on using Resnet50 model. Have you been able to achieve it?

Thanks

raspberry-pi-maker, 2020-05-07:
Hi @IsmaelElHouas .
I didn't test Resnet50. 
I just hope the NVidia experts make the model for us.
I'm only interested in Jetson Edge computing.
Thanks.