NVIDIA-AI-IOT/trt_pose #18: Problem with trt_pose.coco

Kammmil, 2019-12-18:
Hey guys, when trying to run your jupyter demo i have a problem with trt_pose.coco

RuntimeError                 Traceback (most recent call last)
<ipython-input-3-0c2219f9314e> in <module>
      1 import json
----> 2 import trt_pose.coco
      3 
      4 with open('human_pose.json', 'r') as f:
      5     human_pose = json.load(f)

/usr/local/lib/python3.6/dist-packages/trt_pose-0.0.0-py3.6-linux-aarch64.egg/trt_pose/coco.py in <module>
      9 import trt_pose.plugins
     10 import glob
---> 11 import torchvision.transforms.functional as FT
     12 import numpy as np
     13 from trt_pose.parse_objects import ParseObjects

/usr/local/lib/python3.6/dist-packages/torchvision-0.5.0a0+d32bce0-py3.6-linux-aarch64.egg/torchvision/__init__.py in <module>
      1 import warnings
      2 
----> 3 from torchvision import models
      4 from torchvision import datasets
      5 from torchvision import ops

/usr/local/lib/python3.6/dist-packages/torchvision-0.5.0a0+d32bce0-py3.6-linux-aarch64.egg/torchvision/models/__init__.py in <module>
     10 from .shufflenetv2 import *
     11 from . import segmentation
---> 12 from . import detection
     13 from . import video
     14 from . import quantization

/usr/local/lib/python3.6/dist-packages/torchvision-0.5.0a0+d32bce0-py3.6-linux-aarch64.egg/torchvision/models/detection/__init__.py in <module>
----> 1 from .faster_rcnn import *
      2 from .mask_rcnn import *
      3 from .keypoint_rcnn import *

/usr/local/lib/python3.6/dist-packages/torchvision-0.5.0a0+d32bce0-py3.6-linux-aarch64.egg/torchvision/models/detection/faster_rcnn.py in <module>
     11 
     12 from .generalized_rcnn import GeneralizedRCNN
---> 13 from .rpn import AnchorGenerator, RPNHead, RegionProposalNetwork
     14 from .roi_heads import RoIHeads
     15 from .transform import GeneralizedRCNNTransform

/usr/local/lib/python3.6/dist-packages/torchvision-0.5.0a0+d32bce0-py3.6-linux-aarch64.egg/torchvision/models/detection/rpn.py in <module>
      9 from torchvision.ops import boxes as box_ops
     10 
---> 11 from . import _utils as det_utils
     12 from .image_list import ImageList
     13 

/usr/local/lib/python3.6/dist-packages/torchvision-0.5.0a0+d32bce0-py3.6-linux-aarch64.egg/torchvision/models/detection/_utils.py in <module>
     17 
     18 @torch.jit.script
---> 19 class BalancedPositiveNegativeSampler(object):
     20     """
     21     This class samples batches, ensuring that they contain a fixed proportion of positives

~/.local/lib/python3.6/site-packages/torch/jit/__init__.py in script(obj, optimize, _frames_up, _rcb)
   1217         if _rcb is None:
   1218             _rcb = _jit_internal.createResolutionCallback(_frames_up + 1)
-> 1219         _compile_and_register_class(obj, _rcb, qualified_name)
   1220         return obj
   1221     else:

~/.local/lib/python3.6/site-packages/torch/jit/__init__.py in _compile_and_register_class(obj, rcb, qualified_name)
   1074 def _compile_and_register_class(obj, rcb, qualified_name):
   1075     ast = get_jit_class_def(obj, obj.__name__)
-> 1076     _jit_script_class_compile(qualified_name, ast, rcb)
   1077     _add_script_class(obj, qualified_name)
   1078 

RuntimeError: class '__torch__.torchvision.models.detection._utils.BalancedPositiveNegativeSampler' already defined. (register_type at ../torch/csrc/jit/script/compilation_unit.h:166)
frame #0: c10::Error::Error(c10::SourceLocation, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) + 0x78 (0x7f92c938d8 in /home/kamil/.local/lib/python3.6/site-packages/torch/lib/libc10.so)
frame #1: <unknown function> + 0x56c860 (0x7f93294860 in /home/kamil/.local/lib/python3.6/site-packages/torch/lib/libtorch_python.so)
frame #2: <unknown function> + 0x554e3c (0x7f9327ce3c in /home/kamil/.local/lib/python3.6/site-packages/torch/lib/libtorch_python.so)
frame #3: <unknown function> + 0x555b10 (0x7f9327db10 in /home/kamil/.local/lib/python3.6/site-packages/torch/lib/libtorch_python.so)
frame #4: <unknown function> + 0x1ede5c (0x7f92f15e5c in /home/kamil/.local/lib/python3.6/site-packages/torch/lib/libtorch_python.so)
frame #5: _PyCFunction_FastCallDict + 0x1ac (0x5b9b2c in /usr/bin/python)
frame #6: /usr/bin/python() [0x529958]
frame #7: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #8: /usr/bin/python() [0x527860]
frame #9: /usr/bin/python() [0x5297dc]
frame #10: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #11: /usr/bin/python() [0x528ff0]
frame #12: /usr/bin/python() [0x529584]
frame #13: /usr/bin/python() [0x5297dc]
frame #14: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #15: /usr/bin/python() [0x528ff0]
frame #16: /usr/bin/python() [0x5376b4]
frame #17: PyCFunction_Call + 0xd8 (0x5b9ed0 in /usr/bin/python)
frame #18: _PyEval_EvalFrameDefault + 0x718c (0x530fc4 in /usr/bin/python)
frame #19: /usr/bin/python() [0x528ff0]
frame #20: /usr/bin/python() [0x529584]
frame #21: /usr/bin/python() [0x5297dc]
frame #22: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #23: /usr/bin/python() [0x527860]
frame #24: /usr/bin/python() [0x5297dc]
frame #25: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #26: /usr/bin/python() [0x527860]
frame #27: /usr/bin/python() [0x5297dc]
frame #28: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #29: /usr/bin/python() [0x527860]
frame #30: /usr/bin/python() [0x5297dc]
frame #31: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #32: /usr/bin/python() [0x527860]
frame #33: _PyObject_FastCallDict + 0x1a8 (0x6058b8 in /usr/bin/python)
frame #34: _PyObject_CallMethodIdObjArgs + 0xb0 (0x606658 in /usr/bin/python)
frame #35: /usr/bin/python() [0x429144]
frame #36: /usr/bin/python() [0x532f54]
frame #37: PyCFunction_Call + 0xa4 (0x5b9e9c in /usr/bin/python)
frame #38: _PyEval_EvalFrameDefault + 0x718c (0x530fc4 in /usr/bin/python)
frame #39: /usr/bin/python() [0x528ff0]
frame #40: /usr/bin/python() [0x529584]
frame #41: /usr/bin/python() [0x5297dc]
frame #42: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #43: /usr/bin/python() [0x528ff0]
frame #44: _PyFunction_FastCallDict + 0x1d4 (0x529314 in /usr/bin/python)
frame #45: _PyObject_FastCallDict + 0x1a8 (0x6058b8 in /usr/bin/python)
frame #46: _PyObject_CallMethodIdObjArgs + 0xb0 (0x606658 in /usr/bin/python)
frame #47: /usr/bin/python() [0x429248]
frame #48: _PyEval_EvalFrameDefault + 0x6a90 (0x5308c8 in /usr/bin/python)
frame #49: /usr/bin/python() [0x528ff0]
frame #50: /usr/bin/python() [0x5376b4]
frame #51: PyCFunction_Call + 0xd8 (0x5b9ed0 in /usr/bin/python)
frame #52: _PyEval_EvalFrameDefault + 0x718c (0x530fc4 in /usr/bin/python)
frame #53: /usr/bin/python() [0x528ff0]
frame #54: /usr/bin/python() [0x529584]
frame #55: /usr/bin/python() [0x5297dc]
frame #56: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #57: /usr/bin/python() [0x527860]
frame #58: /usr/bin/python() [0x5297dc]
frame #59: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #60: /usr/bin/python() [0x527860]
frame #61: /usr/bin/python() [0x5297dc]
frame #62: _PyEval_EvalFrameDefault + 0x4770 (0x52e5a8 in /usr/bin/python)
frame #63: /usr/bin/python() [0x527860]

Is there any chance you could help me and give me some advice what to do to make it work?

jaybdub, 2019-12-18:
Hi,

Thanks for reaching out!

From what I can tell, looks like there is an issue in torchvision 0.5.  Could you try reverting to 0.4 or 0.3 to see if the problem persists?

Best,
John

Kammmil, 2019-12-19:
Hi John,
thanks for the help,
works like a charm.

Happy Christmas and NYE :)

jaybdub, 2019-12-19:
Great to hear.  Happy holidays!

gooohjy, 2020-05-25:
Hi, I am facing the same problem while trying to run the live_demo.ipnyb file as well. Any idea on how to revert torchvision from 0.5 to 0.4/0.3? I tried running `pip install torch===1.2.0 torchvision===0.4.0 -f https://download.pytorch.org/whl/torch_stable.html` according to the guide from PyTorch [here](https://pytorch.org/get-started/previous-versions/) but the following error popped up:

```
Defaulting to user installation because normal site-packages is not writeable
Looking in links: https://download.pytorch.org/whl/torch_stable.html
ERROR: Could not find a version that satisfies the requirement torchvision===0.4.0 (from versions: 0.1.6, 0.1.7, 0.1.8, 0.1.9, 0.2.0, 0.2.1, 0.2.2, 0.2.2.post2, 0.2.2.post3)
ERROR: No matching distribution found for torchvision===0.4.0
```
Out of all the versions listed in the error message above, none of them were version 0.4/0.3.

I'm currently working on a Jetson Nano and I've installed torch by using the [jetson-inference repository](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md). Any help will be appreciated.