NVIDIA-AI-IOT/trt_pose #21: How to know the peak's body part

luzhangyi319, 2020-01-16:
Is there a way to know what body part a peak point belongs to? I want to know which points are belong to left hand, right hand or left foot etc. Thanks.

Adnaram, 2020-01-30:
```
parts = trt_pose.coco.coco_category_to_parts(human_pose)
```

And this in function draw_object to add the text on the image

```
part = parts[j]
peak = normalized_peaks[0][j][k]
x = round(float(peak[1]) * width)
y = round(float(peak[0]) * height)
cv2.circle(image, (x, y), 3, color, 2)
cv2.putText(image, part, (x + 5, y + 5), cv2.FONT_HERSHEY_SIMPLEX, 1, (0, 0, 255), 1, cv2.LINE_AA)
```

ElegantLin, 2020-07-12:
Hi, @Adnaram , Thanks for your reply. I used your code in my project. However, I found that `part` is a tensor rather than a string. It will throw an error when you call the `cv2.putText` function. Is there something wrong with my code? 

Thanks