NVIDIA-AI-IOT/trt_pose #22: How to run the 256x256 model?

lucyd42, 2020-01-20:
I was able to successfully run the 224x224 model after a couple days of trying. I want to run the 256x256 model provided in the readme. When I download it and try to use it with demo.ipynb, it says "keyError : 'engine' (full output below). 

I suspect this is beacause the new model might have some different json keys/parameters becaus it is different in structure, so my question is how do I run the 256x256 model using the sample code, what all changes will I have to make?

Console Output : 

`Reading TensorRT model.......
Traceback (most recent call last):
  File "/home/lucyd/Work/trackAi/v1.0/trt_pose/tasks/human_pose/live_demo.py", line 35, in <module>
    model_trt.load_state_dict(torch.load(OPTIMIZED_MODEL))
  File "/home/lucyd/.local/lib/python3.6/site-packages/torch/nn/modules/module.py", line 830, in load_state_dict
    load(self)
  File "/home/lucyd/.local/lib/python3.6/site-packages/torch/nn/modules/module.py", line 825, in load
    state_dict, prefix, local_metadata, True, missing_keys, unexpected_keys, error_msgs)
  File "/usr/local/lib/python3.6/dist-packages/torch2trt/torch2trt.py", line 305, in _load_from_state_dict
    engine_bytes = state_dict[prefix + 'engine']
KeyError: 'engine'`

Also I am running this on Ubuntu Laptop machine instead of a Jetson nano. 
Specifications: 
Ubuntu 18.04
Python 3.6
TensorRT 7 (built from source)
Torch 1.2 (pip install)

Also is capturing total num of frames and dividing by inference time for one frame a correct way to calculate FPS?

Regards
Aditya

JuniorKoukponou, 2020-02-18:
Hi,
I have the same error.
Did you find a solution, please ?
Have a good day
Jr

JuniorKoukponou, 2020-02-18:
If you still search for the solution:
You may convert your resnet18_baseline_att_224x224_A_epoch_249.pth intoresnet18_baseline_att_224x224_A_epoch_249_trt.pth by running the first 6 code cells of the (live_demo)[https://github.com/NVIDIA-AI-IOT/trt_pose/blob/master/tasks/human_pose/live_demo.ipynb] notebook.
Thanks
Jr

closed 2020-02-20 by lucyd42