NVIDIA-AI-IOT/trt_pose #26: Cannot replicate performance results

rteslaru, 2020-02-28:
I made a clean JetPack 4.3 card for the specific purpose of testing the performance of this demo.

After installing the pack and running the `live_demo.ipynb` on my Nano, I can't get anywhere the throughput mentioned in the README.

For resnet18, I get 12-14 FPS at best, after TRT optimization.

I made sure to run `jetson_clocks` before, but other than that, I'm out of ideas on what could be the cause of such a large performance gap between my results and yours.

Would appreciate any thoughts on how to reach the same throughput you achieved.

jaybdub, 2020-03-02:
Hi rteslaru,

Have you also set the Jetson Nano to ``10W`` mode?

```bash
sudo nvpmodel -m0
```

Please let me know if this works, otherwise do you mind sharing

1. The version of PyTorch installed
2. The version of torchvision installed

Best,
John

rteslaru, 2020-03-07:
Yes, I have set the Nano to `10W` mode.

PyTorch: 1.4.0
torchvision: 0.2.2

rteslaru, 2020-03-07:
In addition to the version information above, I find strange that the performance does not change at all after jetson_clocks is run. I would have expected the performance to be different.

`sudo nvpmodel -m0`
`sudo jetson_clocks`

```
SOC family:tegra210 Machine:NVIDIA Jetson Nano Developer Kit
Online CPUs: 0-3
CPU Cluster Switching: Disabled
cpu0: Online=1 Governor=schedutil MinFreq=1479000 MaxFreq=1479000 CurrentFreq=1479000 IdleStates: WFI=0 c7=0
cpu1: Online=1 Governor=schedutil MinFreq=1479000 MaxFreq=1479000 CurrentFreq=1479000 IdleStates: WFI=0 c7=0
cpu2: Online=1 Governor=schedutil MinFreq=1479000 MaxFreq=1479000 CurrentFreq=1479000 IdleStates: WFI=0 c7=0
cpu3: Online=1 Governor=schedutil MinFreq=1479000 MaxFreq=1479000 CurrentFreq=1479000 IdleStates: WFI=0 c7=0
GPU MinFreq=921600000 MaxFreq=921600000 CurrentFreq=921600000
EMC MinFreq=204000000 MaxFreq=1600000000 CurrentFreq=1600000000 FreqOverride=1
Fan: speed=255
NV Power Mode: MAXN
```