NVIDIA-AI-IOT/trt_pose #32: How to use model with deepstream

GstreamNovice, 2020-03-15:
Hello, I was curious to see if the pose-estimation model can be used with deepstream. 

It's rather straight forward to convert the model to the model to tensorrt, I am more interested to know what I need to do next.

Do i have to implement the post processing in gstds-example ? How should i go about doing that ?

I know I am not the only one looking for an answer to this question. So any help would be greatly appreciated by me and by the community.

jaybdub, 2020-03-15:
Hi GstreamNovice,

Thanks for reaching out!  

Do you mind sharing a reference to the DeepStream sample you're basing off of?  I am less familiar in this area but will try to help how I can.

As for this project, there are a few stages

1.  Neural network execution  (GPU, TensorRT).  This takes one input binding (the image) and produces two output bindings (the confidence map and part affinity field).
2.  Post processing (CPU).  This takes the confidence map and part affinity fields (copied into CPU memory), and produces the object counts, object part mappings, and object part coordinates.

Past that, it depends on the application.  In the current examples, both the neural network execution and post processing depend on PyTorch for bindings.  That said, I've taken steps to remove this dependency, which may simplify integrating in an application.

Please let me know if you have any questions, or are able to share your use case so I better understand the challenge your facing.

Best,
John

ishang3, 2020-10-11:
@GstreamNovice I am working on this same problem, were you successful?

thancaocuong, 2020-11-09:
@ishang3  first you need to follow the deepstream apps, for convinience, I think you need to write inference code in C++. It's like Pose plugin for deepstream. Then write the gstream plugin like yolo. So what's your problem?

ishang3, 2020-11-09:
@thancaocuong the datastructure for trt pose is very different - it is not object detection which has the bounding box coordinates, but it contains 18 keypoints. Also, the preprocessing is a little different too where nvidia has not been transparent in what changes need to be done.

thancaocuong, 2020-11-09:
I did implement the preprocess on cuda including convert to RGB,  resize,  normalize.  I will reoganize my code then share with you.  Hope it will help

ishang3, 2020-11-09:
@thancaocuong Thank you, I appreciate it very much.

thancaocuong, 2020-11-10:
@ishang3 please take a look at nvPreprocess function. I use cuda to normalize RGB image and DMA it to GPU for inference. You also can use my repo for trt_pose, but you need to add post-processing function written in c++ (trt_pose repo).  Feel free to ask me if you have  any question. https://github.com/thancaocuong/OpenPifPaf-TensorRT/blob/dev/src/pifpafnet.cpp

zhink, 2020-11-17:
This is example : https://github.com/NVIDIA-AI-IOT/deepstream_pose_estimation