NVIDIA-AI-IOT/trt_pose #38: ModuleNotFoundError: No module named 'trt_pose'

imranmu, 2020-04-11:
- I have successfully installed and build all packages but running the example from Jupyter notebook gives error
__----> 2 import trt_pose.coco
      3 
      4 with open('human_pose.json', 'r') as f:
      5     human_pose = json.load(f)
ModuleNotFoundError: No module named 'trt_pose'__

- I am running it from virtual environment
(venvJetssonPoseEstimation) user@user-desktop:~/workspaceJetson/poseEstimation/trt_pose$

- I have these packages
  Cython (0.29.16)
  numpy (1.18.2)
  Pillow (7.1.1) 
  pip (9.0.1)
  setuptools (46.1.3)
  six (1.14.0)
  torch (1.3.0)
  torch2trt (0.0.3)
  torchvision (0.4.2)
  trt-pose (0.0.1)
  wheel (0.34.2)

I could't see any available solution and tried some related problem's solution.

polya-xue, 2020-06-01:
same problem

jsynnott, 2020-06-23:
Anyone have any success with this? Same problem.

jasonakon, 2020-07-17:
please perform `sudo python setup.py install` and restart your system and try again

anselanza, 2020-09-24:
Got the same error, worked for me after `sudo python setup.py install` from within the repo root.

TheoLeRlgolo, 2025-07-08:
> Got the same error, worked for me after `sudo python setup.py install` from within the repo root.

I doo al these from my sistem terminal or from jupiterlab dlinano ?