NVIDIA-AI-IOT/trt_pose #39: How can I train trt_pose on multi-GPU?

kinglintianxia, 2020-04-23:
@jaybdub
Hi, jaybdub, Thanks for your great work! 
When I was trainning the net, I find the trainning process on only one GPU(I have two),  how can I train trt_pose on multi-GPU? And restore from a pretrained checkpoint?

AllenDun, 2020-04-24:
this repo not support multi-gpu trainning now?

kinglintianxia, 2020-04-25:
> this repo not support multi-gpu trainning now?
I used to train on a  machine with two 1080ti, only gpu: 0 used in the training, have you tried it?

kinglintianxia, 2020-05-19:
Finally I solved this.

closed 2020-05-19 by kinglintianxia

jpapon, 2021-10-21:
For reference, I just added:
`model = torch.nn.DataParallel(model)`
after the amp initialization in `train.py` and it uses multiple gpus.
It looks like I'm not constrained by the data loader though, so I don't think that's being parallelized properly.
@kinglintianxia did you do anything else for this?