NVIDIA-AI-IOT/trt_pose #40: No module named 'trt_pose.plugins'

adriati, 2020-04-29:
When I run it on Jupyter, it works but when I try to run it in PyCharm, with the same Python interpreter as the Jupyter, it returns to me this error. Someone knows why?

Thanks in advance

nakulkhajuria, 2020-05-08:
same error i am also getting, while i try to run it in PyCharm. plus, iam also getting problem in running this project on jupyter. can u help me with it. @adriati

Param-Raval, 2020-05-14:
same issue in Google Colab

hfnvbh, 2020-06-09:
Same issue in Google colab. Don't get how to run it properly, since in colab it's only one notebook and manipulations with paths are tricky.

Sifat-Ahmed, 2020-07-22:
Does anyone have a solve to this issue?

ljmd21, 2020-08-18:
Having the same problem here.... Anyone found a fix?

Sifat-Ahmed, 2020-08-18:
> Having the same problem here.... Anyone found a fix?

Yes. One solve is that after installing don't run the code here. If you import it somewhere else, on another directory it will work. There is a trt_pose folder which contains plugins.cpp. If your root directory contains this trt_pose folder it will not work because interpreter gets confused.

Another way is to after installing you will see an .egg file. Inside that .egg file you will find a trt_pose folder where there is a plugins.py file. you can replace this trt_pose folder with the trt_pose folder from egg file.

The main problem is

trt_pose
--tasks/human_pose
-- trt_pose (***)
--CMakeList.txt

the trt_pose folder with (***) causes all the problems. You can replace folder from the .egg trt_pose folder or you can try to import trt_pose.plugins in a different directory. Hope it'll work.

Sifat-Ahmed, 2023-03-06:
> @Sifat-Ahmed I can't find the plugins.py file inside that .egg file in the folder trt_pose
> 
> i jut have the plugins.cpp

Please mention your system details and things you have tried so that I reproduce the issue and help you.