NVIDIA-AI-IOT/trt_pose #44: FPS on Jetson Xavier

IsmaelElHouas, 2020-05-05:
Hi!
I achieved to run Resnet18 on Jetson Xavier with a result of 30 fps.
How did you get the performance of 251 fps?

Thanks

thancaocuong, 2020-05-19:
251 FPS is only time consuming for inference process (does not consist of postprocessing process). Anyway, I obtained overall 7FPS when  running  densenet121 320x320 on jetson tx2 !

thancaocuong, 2020-05-19:
hi @IsmaelElHouas, 251 FPS you got is only inference time (without postprocessing time called paf process). Anyway, I obtained overall 7fps when running densenet121 320x320 on jetson tx2. So on Xavier, It should be better.

Tetsujinfr, 2020-06-02:
Fyi, I got 128fps on a Xavier NX, so I guess 251fps on a Xavier AGX makes sense. I have used the 2 cores/15W mode, i.e. dedicating as much power envelope to the GPU as possible. E.g. if I select the 4cores/15W power mode then the fps drops to 82.
Just in case you did not consider the power modes carefully.

jasonakon, 2020-07-20:
@IsmaelElHouas , @thancaocuong , may i know which version of pytorch and torchvision you installed to run the inference ? I always met this issue all the time. 
[image]

thancaocuong, 2020-07-20:
@jasonakon  you need to remove trt_pose completely and reinstall.  On jetsonNX, and Jetson TX2 I use torch1.5.0 with jetpack 4.4. So you can try to use lastest jetpack version. I use it without any problems