NVIDIA-AI-IOT/trt_pose #50: Several Problems with live_demo.ipynb

BowenGIT, 2020-05-26:
I recently tried running the live_demo.ipynb, but the camera resolution is extremely weird, and I can only init the camera once before I have to restart it. Twice, and there is an error. When the camera does run, there is no output. It is just a video feed. There is no pose detection. Help please!

uersoy, 2020-05-31:
I had the same issue.

gohjiayi, 2020-06-16:
For me, I was unable to get a video feed when I run the cell that was supposed to create a widget used to display the camera feed with visualizations as seen below.

```
import ipywidgets
from IPython.display import display

image_w = ipywidgets.Image(format='jpeg')

display(image_w)
```

It gives me a broken image icon. FYI I'm working on a CSI camera and I have adjusted the code before accordingly. Camera is working well on other programs as well. Any idea why?

curiosity-rover, 2020-06-30:
@gohjiayi perhaps this link will help ypu:
https://stackoverflow.com/questions/36351109/ipython-notebook-ipywidgets-does-not-show

basically, you will need to enable the extension using `jupyter nbextension enable --py widgetsnbextension` in the terminal window. Hope this helps!

tomMEM, 2020-07-10:
Hello, I also got the broken  "image icon"  after cell 13 at first, but just going on with executing  cell 14 and 15 and 16 was enough,
Works nicely, just a delay of 2 to 10 seconds, might be because of WIFI connection.