NVIDIA-AI-IOT/trt_pose #55: Error trt_pose.coco not found

jayajohnson, 2020-07-08:
I am trying to install I ran the following commands in google colab
 % cd /trt_pose/
!sudo python3 setup.py install

It seems to install the library as trt_pose.trt_pose

So gives me the error trt_pose.coco not found.

Daniel-10, 2020-08-05:
Hmm it may be this problem.  I'll give this a shot right quick.

https://github.com/ClementPinard/Pytorch-Correlation-extension/issues/15

jaybdub, 2020-08-06:
Hi Jayajohnson,

Thanks for reaching out!

My initial guess is that you may be running this from the root of the ``trt_pose`` project directory.

If this is the case, your python program may be attempting to load the compiled extensions from the local folder rather than from the installed package.  To do this, you need to build the extensions in place

```bash
python3 setup.py build_ext --inplace
```

Please let me know if this helps or you run into any issues.

Best,
John

wwdok, 2020-12-04:
@ jaybdub my working directory is not in the root of the trt_pose project directory ,but i  still have the issue:
[image]
and i have already run `python3 setup.py build_ext --inplace`
[image]

zeta0707, 2020-12-26:
@ jaybdub
I also got same problem, no resolution yet.

[image]

And i have already run python3 setup.py build_ext --inplace

My environment:
Jetpack4.4.1(=Jetbot0.4.2)
```
jetbot@nano-4gb-jp441:~/trt_pose$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
>>> import torch
>>> print(torch.__version__)
1.7.0
>>> import torchvision
>>> print(torchvision.__version__)
0.8.0a0+45f960c
```

zeta0707, 2020-12-26:
I downgraded torch and torchvision, though I see the same problem.

```
jetbot@nano-4gb-jp441:~/trt_pose$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> print(torch.__version__)
1.6.0
>>> import torchvision
>>> print(torchvision.__version__)
0.7.0a0+78ed10c
```

zeta0707, 2020-12-26:
I ran below code for checking what happens.
Why there's undefined symbol?

```
jetbot@nano-4gb-jp441:~/torch2trt$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import trt_pose.coco
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-aarch64.egg/trt_pose/coco.py", line 9, in <module>
    import trt_pose.plugins
ImportError: /usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-aarch64.egg/trt_pose/plugins.cpython-36m-aarch64-linux-gnu.so: undefined symbol: **_ZN3c104impl23ExcludeDispatchKeyGuardC1ENS_14DispatchKeySetE**
```

Hiroaki-K4, 2021-04-22:
@zeta0707 
Hi, Have you solved?
I have a same problem.

zeta0707, 2021-04-23:
Unfortunately not yet.

Thanks

Hiroaki-K4, 2021-04-23:
ok.
In my case, It was fixed after replacing with the latest trt_pose.

zeta0707, 2021-04-23:
Hi

Great, what version of trt_pose?
I'm afraid that sometimes latest misreading.

attiasa22, 2021-06-30:
Hi,

Has a fix for this issue been found?

MuhammadMateenShahid, 2023-09-28:
I am also facing this issue. Has anyone found the solution?

ysun, 2024-03-03:
I have the same issue, any solution ?