NVIDIA-AI-IOT/trt_pose #56: trt_pose not utilizing the GPU on jetson Tx2

AmoghBharadwaj, 2020-07-22:
Hello All,

We are trying to develop an application which uses trt_pose, the model is too slow and the max we could extract was 15fps on Jetson Tx2, after nvpmodel and jetson_clocks. Also, I observed that through tegrastats, the code is not utilizing all the cores, and I believe the GPU cores have zero usage during runtime, even though it is enabled inside the code. Can you please help me in getting a boost in the performance? Thank you.

jaybdub, 2020-07-22:
Hi AmoghBharadwaj,

Thanks for reaching out!

Do you mind sharing the code you used to optimize and execute the model?  Could you also share

1. JetPack version
2. PyTorch version
3. TensorRT version
4. torchvision version

Please let me know if you have any questions.

Best,
John

AmoghBharadwaj, 2020-07-23:
Hello John, Thank you so much for replying.

I am attaching the code for your reference. 

Jetpack version : 4.3DP
Pytorch version : 1.5.0
TensorRT Version : 7.1.3
Torchvision version : 0.6.0

Regards,
Amogh Bharadwaj
[trt_pose.txt](https://github.com/NVIDIA-AI-IOT/trt_pose/files/4963323/trt_pose.txt)

AmoghBharadwaj, 2020-07-23:
Hello John, also attaching the tegrastatus window capture for your reference. 
the nvpmodel is in mode 0. 
[image]

jaybdub, 2020-07-23:
Hi AmoghBharadwaj,

Thanks for sharing.  I just took a brief look.

Does the timing that you witness include the OpenCV visualization?

What is the timing reported by just this section?

```python
model_trt = TRTModule()
model_trt.load_state_dict(torch.load(OPTIMIZED_MODEL))

t0 = time.time()
torch.cuda.current_stream().synchronize()
for i in range(50):
    y = model_trt(data)
torch.cuda.current_stream().synchronize()
t1 = time.time()

print(50.0 / (t1 - t0))
```

Best,
John

AmoghBharadwaj, 2020-08-03:
Hello @jaybdub , sorry for the late reply. 

The model is running fine and is using GPU power, I am able to run it near real time. I made a mistake while seeing the console window and thought it was not utilising gpu. All good now, Thank you so much for the support

closed 2020-08-03 by AmoghBharadwaj