NVIDIA-AI-IOT/trt_pose #57: can we deploy it in deepstream? if yes, how to deploy it?

shuxuzhipeng, 2020-07-27:


ishang3, 2020-08-06:
I am interested in the same problem, were you able to see any helpful resources ?

shuxuzhipeng, 2020-08-10:
> I am interested in the same problem, were you able to see any helpful resources ?

1 hyperpose
it also use tensorrt to do the inference and its model file is the format : onnx file. So that it can be deployed in deepstrem.
2 CMU-Perceptual-Computing-Lab/openpose
its model file is the format, caffemode which also could be used in deepstream.
But u should define the post process customer function to parser the result of inference.
Hope u good luck

Y07yoyo, 2020-08-26:
Hi @shuxuzhipeng,

Are you able to provide a bit more information on how to use the openpose caffemodel into deepstream?

Kind Regards,

ishang3, 2020-09-24:
@Y07yoyo I am solving the same issue - were you successful?

thancaocuong, 2020-11-11:
@shuxuzhipeng following this example with custorm [yolo](https://github.com/AlexeyAB/deepstream-plugins), we can write the poseplugin to integrate with deepstream.

zhink, 2020-11-17:
deepstream example in here: https://github.com/NVIDIA-AI-IOT/deepstream_pose_estimation