NVIDIA-AI-IOT/trt_pose #58: model post-processing modify

sunbin1357, 2020-07-28:
I want to use your provided c++ code for model post-processing, but my keypoint is 19 and paf is 38, What parts of your provided c++ code  do I need to modify.

jaybdub, 2020-08-03:
Hi sunbin1357,

Thanks for reaching out!

You should not need to modify the functions directly, but just provide different arguments to match your configuration.

There are some more notes here

https://github.com/NVIDIA-AI-IOT/trt_pose/tree/master/trt_pose/parse

I'd get it to work in Python first, using the bindings for these functions.

You can see how these methods are called here

https://github.com/NVIDIA-AI-IOT/trt_pose/blob/master/trt_pose/parse_objects.py#L17

The toplology can be generated from the COCO annotation 

https://github.com/NVIDIA-AI-IOT/trt_pose/blob/master/trt_pose/coco.py#L21

Please let me know if this helps or you run into any issues.

Best,
John

sunbin1357, 2020-08-05:
According to your suggestion, I modified the corresponding code, but the
result is shown in the below

thancaocuong, 2020-08-10:
@sunbin1357 can you show me your modifications? I will check it out and give you some suggestions

sunbin1357, 2020-08-10:
I add tensor.contiguous(), and the problem is solved