NVIDIA-AI-IOT/trt_pose #61: Import Error

Daniel-10, 2020-08-05:
Hello,  Has anyone happen to have gotten this error when importing trt_pose.coco before?

---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
 in 
      1 import json
----> 2 import trt_pose.coco
      3 
      4 with open('human_pose.json', 'r') as f:
      5     human_pose = json.load(f)

/usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-aarch64.egg/trt_pose/coco.py in 
      7 import tqdm
      8 import trt_pose
----> 9 import trt_pose.plugins
     10 import glob
     11 import torchvision.transforms.functional as FT

ImportError: /usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-aarch64.egg/trt_pose/plugins.cpython-36m-aarch64-linux-gnu.so: undefined symbol: _ZTIN3c1021AutogradMetaInterfaceE

Daniel-10, 2020-08-05:
Hmm it may be this problem. I'll give this a shot right quick.

ClementPinard/Pytorch-Correlation-extension#15

Daniel-10, 2020-08-05:
This worked like a champ.  I deleted the build folders and just rebuilt and all is good.

closed 2020-08-05 by Daniel-10

md-bd, 2020-08-06:
Hello, I am trying to run it and found the same error.
Traceback (most recent call last):
  File "demo.py", line 27, in <module>
    import trt_pose.coco
  File "/usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-x86_64.egg/trt_pose/coco.py", line 9, in <module>
    import trt_pose.plugins
ImportError: /usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-x86_64.egg/trt_pose/plugins.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceISt7complexIfEEEPKNS_6detail12TypeMetaDataEv

Do I delete /usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-x86_64.egg/ folder and then reinstall just trt_pose again? Or do i have do reinstall a lot of stuff?

Daniel-10, 2020-08-06:
Hello md-bd,  I only needed to delete the trt pose folder and then redownload/rebuild/reinstall just that.