NVIDIA-AI-IOT/trt_pose #67: what is the 'person_keypoints_val2017_modified.json'?

zzwolf, 2020-09-09:
hello, guys.

I am trying to validate the pre-trained model provided by the author, but when I execute tasks/human_pose/eval.ipynb, it is said that I do not have 'person_keypoints_val2017_modified.json'.

I noticed that instead of 17 keypoints, the author use 18 keypoints, perhaps the 'NECK' point was calculated automatically.

So I was wondering how to convert the official 'person_keypoints_val2017.json' to 'person_keypoints_val2017_modified.json'.

Thanks!

jaybdub, 2020-09-13:
Hi zzwolf,

Thanks for reaching out!

That's correct, we automatically add the 'neck' key point as the midpoint between the shoulders.

To create the modified annotations, you can run this script

https://github.com/NVIDIA-AI-IOT/trt_pose/blob/master/tasks/human_pose/preprocess_coco_person.py

Please let me know if this helps or you run into any issues.

Best,
John

zzwolf, 2020-09-19:
> Hi zzwolf,
>
> Thanks for reaching out!
>
> That's correct, we automatically add the 'neck' key point as the midpoint between the shoulders.
>
> To create the modified annotations, you can run this script [...]

Great! I successfully convert 'person_keypoints_val2017.json' to 'person_keypoints_val2017_modified.json'. Thanks a lot.

However when I ran this demo on jetson nano, I only got 10FPS by using resnet18_baseline_att_224x224_A_trt.pth. Do you know any reason why so and any particular place I need to pay attention?

Best wishes
zzwolf