NVIDIA-AI-IOT/trt_pose #68: Execution on normal ".py" file type

ApluUalberta, 2020-09-10:
Hi there,
Thank you for the amazing library! I am currently trying to test the library on the Jetson Xavier NX, but the script in the live_demo.ipynb script's following code seems to only be meant for jupyter ipynb, which I am trying to avoid using currently:

`import ipywidgets`
`from IPython.display import display`

`image_w = ipywidgets.Image(format='jpeg')`

`display(image_w)`

I have tried to use opencv as an alternative, with limited success (running and displaying but not drawing):

`cap = cv2.VideoCapture(0)`
`while true:`
`    ret, frame = cap.read()`
`    execute({'new' : frame})`
`    cv2.imshow('Pose Estimation', frame)`
`    if cv2.waitKey(1) & 0xFF ==ord('q'):`
`        break`

`cap.release()`
`cv2.destroyAllWindows()`

I was wondering if anyone else has gotten this running with just a normal .py script. Any library recommendations or something i'm missing?

ljmd21, 2020-09-14:
Hi! I'm wondering the exact same thing, although it seems like you've gotten way further than me, I've identified ipywidgets as the issue to resolve for using and displaying the live camera stream but don't know how to do it. Any help would be greatly appreciated!

Hasankanso, 2020-09-25:
Hello, I'm encountering the same problem, although I've made it to display opencv inside of jupyter using follwing steps:

replace
```
from jetcam.usb_camera import USBCamera
# from jetcam.csi_camera import CSICamera
from jetcam.utils import bgr8_to_jpeg

camera = USBCamera(caputre_device=1)
# camera = CSICamera(width=WIDTH, height=HEIGHT, capture_fps=30)

camera.running = True
```
with:
`cap1 = cv2.VideoCapture(0)`

replace execute method section with:

```
from jetcam.utils import bgr8_to_jpeg

def execute(image):
    data = preprocess(image)
    cmap, paf = model_trt(data)
    cmap, paf = cmap.detach().cpu(), paf.detach().cpu()
    counts, objects, peaks = parse_objects(cmap, paf)#, cmap_threshold=0.15, link_threshold=0.15)
    draw_objects(image, counts, objects, peaks)
    image_w.value = bgr8_to_jpeg(image[:, ::-1, :])
```

finally replace `execute({'new': camera.value})` with:

```
while True:
     ret,frame = cap.read()
     execute(frame)

cap.release()
cv2.destroyAllWindows()
```
but I only get the cam image without any addition, when I print `counts ` inside of execute method I get:

**tensor([0], dtype=torch.int32)**

it's not finding anything? probably there's something wrong with the image from preprocess(image), meaning the image is not compatible with preprocess method.
but I'm not sure anyway

Hasankanso, 2020-09-28:
Hello, just resize the image to 224x224 before passing it to the execute method and it should work, by the way you can open live_demo in jupyter and save a copy as .py version then you can run it as normal python (python live_demo.py)
check how I fixed it( change format to ipynb before):
https://github.com/NVIDIA-AI-IOT/trt_pose/files/5289383/live_demo.txt

Edohvin, 2020-10-19:
> Hi there,
> Thank you for the amazing library! I am currently trying to test the library on the Jetson Xavier NX, but the script in the live_demo.ipynb script's following code seems to only be meant [...]

Have you been able to fix it? Hasankanso's solution doesn't seem to work for jetson devices.

Funderburger, 2021-05-21:
Just put a `cv2.imshow("Skeleton",image)` at the end of the `execute` function created by Hasankanso. In my case it worked on Jetson NX.

mjysci, 2021-06-25:
[This](https://github.com/NVIDIA-AI-IOT/trt_pose/pull/4) may help.