NVIDIA-AI-IOT/trt_pose #71: run setup.py segmentation fault

SunshineLiZhen, 2020-09-22:
hello,I use "sudo python3 setup.py  install"  to setup trt_pose. but ,then it stopped. on the screen,it only show me "segmentation fault".
      I do not know what's wrong with it. please help me!

sureshbadagi, 2020-09-22:
Hi, I am finding same problem by running sudo python3 setup.py  install command, please help.

SunshineLiZhen, 2020-09-25:
I found thar，when you import torch ,it show  "segmentation fault" .so the problem is pytorch.

riyan-datalyca, 2021-05-31:
Im having the same issue. I installed pytorch from  [nvidia forums](https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-8-0-now-available/72048)  version 1.7 in my jetson nano. When I call `import torch` there is no error like `segmentation falut`. Is there any fix for this issue?