NVIDIA-AI-IOT/trt_pose #73: replacing jetcam with opencv works but only shows image without any drawing on it.

Hasankanso, 2020-09-27:
Hello everyone, 

this issue is redundant, I re-post it due the misleading title in the [original one #68](https://github.com/NVIDIA-AI-IOT/trt_pose/issues/68)

this is the live_demo with opencv, just change the the file format to ipynb (github doesn't allow ipynb files attached):
[live_demo.txt](https://github.com/NVIDIA-AI-IOT/trt_pose/files/5289229/live_demo.txt)

I'm feeding the image to the execute method, but the count of people is always 0(which is wrong).

Hasankanso, 2020-09-28:
the opencv image was not resized to 224x224, did it and it works now.
[live_demo.txt](https://github.com/NVIDIA-AI-IOT/trt_pose/files/5289383/live_demo.txt)

closed 2020-09-28 by Hasankanso

yuerunyu, 2020-09-29:
@Hasankanso hello,I used your live_demo.txt,change format to ipynb,but there is no result,no windows,no image.when execute "display(image_w)",output is "Image(value=b'', format='jpeg')",and last code "while true:",there is no respond.can you help me ?my device is jetson nano.

Hasankanso, 2020-09-30:
Hello,
I'm using linux on laptop as an environment for the trt_pose, I don't have any nvidia device beside the graphic card ofcourse. I'm not sure what is the environment of jetson nano, thus I can't help you further. Maybe I can help you, if your provide more information about the platform. However the while true loop is blocking the instance, maybe you should put what is inside of the loop in a function and pass it as a callback function of cap.read(), I'm not sure if this is possible either. Sorry for not being able to help more.

Edohvin, 2020-10-19:
@yuerunyu Have you been able to fix it for jetson nano?

yuerunyu, 2020-10-20:
> @yuerunyu Have you been able to fix it for jetson nano?

No,I have not fix it for jetson nano,sorry..

yuerunyu, 2020-10-20:
>@Edohvin no,I have not fix it for nano,sorry.