NVIDIA-AI-IOT/trt_pose #74: torch2trt installation

AffDk, 2020-10-07:
Hi,

When I try to install torch2trt, I faced with a number of error and the installation will fail at the end.
it is a long list of errors so I cannot paste the entire message here but I picked up one or two of the error messages from top of the list.
please see below.

BR
Aff
for example:

torch2trt/plugins/interpolate.cpp:51:43: error: no matching function for call to ‘torch::serialize::InputArchive::read(const char [5], c10::IValue&)’
           input_archive.read("size", value);

or

/usr/local/lib/python3.6/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:55:8: note:   no known conversion for argument 2 from ‘c10::IValue’ to ‘torch::serialize::InputArchive&’
torch2trt/plugins/interpolate.cpp:53:39: error: ‘const class std::vector<long int>’ has no member named ‘vec’; did you mean ‘vector’?
           size = value.toIntListRef().vec();

varhidibence, 2020-10-13:
Same problem here: 
[image]

@jaybdub please could you help us?

Thanks

guijuzhejiang, 2020-10-21:
It may be that the version of pytorch is less than 1.3, so the plugin cannot be compiled. You can try the command not to compile the plugin：python setup.py install

varhidibence, 2020-11-04:
> It may be that the version of pytorch is less than 1.3, so the plugin cannot be compiled. You can try the command not to compile the plugin：python setup.py install

Thanks @guijuzhejiang , this solved my installation problem! :)