NVIDIA-AI-IOT/trt_pose #76: Getting pose metadata

rshigemura, 2020-10-13:
Hi @jaybdub! 

Simply awesome the way we can detect poses and plot it on vídeo!! Great job!!

I just couldn't find out how to get pose's features metadata programatically in python (eg.: left_eye, left_elbow, right_ankle, etc.)

Could you please clarify it?

Thanks!

ishangupta3, 2020-10-13:
Hello @rshigemura this blog is going to help you:

https://spyjetson.blogspot.com/2019/12/jetsonnano-human-pose-estimation-using.html

rshigemura, 2020-10-15:
Hello @ishangupta3 !!

Thanks, it was exactly what I was looking for!

closed 2020-10-15 by rshigemura

arvindchandel, 2020-12-17:
Hi @ishangupta3 @rshigemura 
 I am trying to run the model on Nvidia GPU 1650, so while running notebook and cell-5, the code below
import torch2trt
model_trt = torch2trt.torch2trt(model, [data], fp16_mode=False, max_workspace_size=1<<25)

I am getting run time error like below: How i can get rid of this
RuntimeError Traceback (most recent call last)
in
1 import torch2trt
2
----> 3 model_trt = torch2trt.torch2trt(model, [data], fp16_mode=False, max_workspace_size=1<<25)

~/tlt-experiments/skeleton-env/lib/python3.6/site-packages/torch2trt-0.1.0-py3.6-linux-x86_64.egg/torch2trt/torch2trt.py in torch2trt(module, inputs, input_names, output_names, log_level, max_batch_size, fp16_mode, max_workspace_size, strict_type_constraints, keep_network, int8_mode, int8_calib_dataset, int8_calib_algorithm, int8_calib_batch_size, use_onnx)
553 )
554
--> 555 engine = builder.build_cuda_engine(network)
556
557 module_trt = TRTModule(engine, input_names, output_names)

RuntimeError: NVRTC error: