NVIDIA-AI-IOT/trt_pose #77: AttributeError: 'FloatProgress' object has no attribute 'style'

guillebot, 2020-10-15:
On a freshly installed TX2, following the instructions several times, allways get this on the live_demo.ipynb.

Please help.

thanks

closed 2020-10-15 by guillebot

xoxoxo13102020, 2020-11-11:
Did you find any solution to this err?

xoxoxo13102020, 2020-11-11:
`from tqdm import tqdm as tqdm`

this worked for me

MuminKhan, 2020-11-19:
I'm having this same issue, even with the import. @guillebot and @xoxoxo13102020, do you have any other insights?

dzf0023, 2020-11-19:
i got this while i want to do "model = BertForSequenceClassification.from_pretrained('bert-base-uncased')" anyone know how to fix?

guillebot, 2020-11-19:
> I'm having this same issue, even with the import. @guillebot and @xoxoxo13102020, do you have any other insights?

Yes, it is a python library that's missing. Do you have all the dependencies? Matplotlib? I don't remember exactly but fixed installing more libs.

MuminKhan, 2020-11-19:
If you have a working environment, could you possibly run `pip3 freeze` so I can see what's missing

MuminKhan, 2020-11-21:
If you have a working environment, could you possibly run `pip3 freeze` so I can see what's missing @guillebot

swkim01, 2020-12-01:
try to upgrade ipywidgets
`pip install --upgrade ipywidgets`

pablosilgorl, 2021-03-04:
The before mentioned didn't work for me. 
I downgraded tqdm.
`pip3 install tqdm=4.40.0`

ReluXingZeng, 2021-03-04:
^ but `pip3 install tqdm==4.40.0`

pablosilgorl, 2021-03-05:
> ^ but `pip3 install tqdm==4.40.0`

Yes. Thank you.

Xiaobaichipi, 2023-09-20:
[Code:
trans = transforms.ToTensor()
mnist_train = torchvision.datasets.FashionMNIST(
    root="../data", train=True, transform=trans, download=True)
mnist_test = torchvision.datasets.FashionMNIST(
    root="../data", train=False, transform=trans, download=True)]
AttributeError: 'FloatProgress' object has no attribute 'style'
Could anyone tell me how can I work out this error?