NVIDIA-AI-IOT/trt_pose #78: Gstreamer warning: Cannot query video position: status=0, value=-1, duration=-1

josh-github1, 2020-10-15:
Hey everyone,

We're currently trying to run trt_pose on a Jetson TX2 running on Ubuntu 18 with Python3. We did some debugging with different module installations using pip; had to reinstall modules like 'traitlets', 'ipywidgets', etc. Python 3.6; and moved files like human_pose.json and *.pth files for model weights to '~/torch2trt/trt_pose'. We can't seem to get the actual camera to give us live feed. Here's the error:

#from jetcam.usb_camera import USBCamera
... from jetcam.csi_camera import CSICamera
>>> from jetcam.utils import bgr8_to_jpeg
>>>
>>> #camera = USBCamera(width=WIDTH, height=HEIGHT, capture_fps=30)
... camera = CSICamera(width=WIDTH, height=HEIGHT, capture_fps=30)
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 2592 x 1944 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 16.000000; Exposure Range min 34000, max 550385000;

GST_ARGUS: 2592 x 1458 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 16.000000; Exposure Range min 34000, max 550385000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 16.000000; Exposure Range min 22000, max 358733000;

GST_ARGUS: Running with following settings:
   Camera index = 0
   Camera mode  = 2
   Output Stream W = 1280 H = 720
   seconds to Run    = 0
   Frame Rate = 120.000005
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1

No display is coming up and the whole code works fine. The onboard camera doesn't show any output.

Any help would be really appreciated!!

varhidibence, 2020-10-16:
Hi,

Try to restart camera service before each start of the camera:
`sudo systemctl restart nvargus-daemon`

On my Jetson Nano this solves the problem I had.
https://github.com/NVIDIA-AI-IOT/jetbot/issues/35#issuecomment-483405749

bigwhitebird29, 2021-05-08:
posted oct 2020, and still no resolve? not looking good for me then, I've been trying for days to find a resolve. nothing.

3111395644, 2023-07-12:
This problem also raised  to me,but it does not affect my normal use。

3111395644, 2023-07-12:
If you run into question with this log, this may not be the key point,try check somewhere else