NVIDIA-AI-IOT/trt_pose #82: How to use this for identifying Poses??

SijinJohn, 2020-11-09:
Can we use this to identify different poses??like sitting, standing, etc.??
How do we do that??

arvindchandel, 2020-12-18:
Hi @SijinJohn
I recently explored this model and from keypoints, i ma also interested in identifying poses like u mentioned. So did u get any proper method to approach it.

SijinJohn, 2020-12-21:
No i didn't get for this  trt_pose.
But i got one with OpenPose :-
https://github.com/syedumaidahmed/pose-detection-sitting-standing.

if you get for trt_pose pls do mention here.

arvindchandel, 2020-12-21:
yes i too find for OpenPose:
https://github.com/dekucheng/Skeleton-Based-Human-Action-Recognition

arvindchandel, 2020-12-21:
did you use that one to train on some custom activity data.

SijinJohn, 2020-12-21:
No i didnt try on custom data. 
I also don't have much understanding on how to trian for custom dataset.

arvindchandel, 2020-12-21:
I want to try, but issue is that i have to integrate it with deepstream, for that i am looking for a way.

SijinJohn, 2020-12-21:
if you find any solution pls do update here.

tucachmo2202, 2020-12-29:
> I want to try, but issue is that i have to integrate it with deepstream, for that i am looking for a way.

I am looking for too. Please let me know if you have any solutions. Tks!

c7934597, 2021-01-28:
Hi all, I also need that, I am finding about "the pose of using phone".

I also have to integrate it with deepstream on edge device.

Thank you!

arvindchandel, 2021-01-30:
@c7934597 trt_pose with deepstream is not good combination, i tried that, better you try something else.

c7934597, 2021-01-30:
Hi @arvindchandel

https://developer.nvidia.com/blog/creating-a-human-pose-estimation-application-with-deepstream-sdk/

https://github.com/hafizas101/Real-time-human-pose-estimation-and-classification 

I saw these article before. It's good combination? I don't sure that. Maybe I will try it, and then using classification method for solving my need.

arvindchandel, 2021-01-31:
Hi @c7934597 
yes i tried same and re-implemented it in python, but was not satisfied with pose quality so i am trying other state of art, but u can try for your use case, may be it fits in your requirement.

c7934597, 2021-01-31:
Hi @arvindchandel 
Other state of art, example what?
I also wanna understand those. Thank you.

tucachmo2202, 2021-02-01:
Hi @arvindchandel, 
I am also interested in this problem. I see that the quality of this pose app not good enough to estimation action but I haven't found any solution to for realtime problem. Can you help me? 
Thanks so much

dreamilk, 2021-03-05:
Hi all, I also need that.Have you come up with a good solution? Can you share it？ @c7934597 @arvindchandel @SijinJohn @tucachmo2202 @jaybdub

tucachmo2202, 2021-03-05:
@dreamilk you can try this article https://towardsdatascience.com/fall-detection-using-pose-estimation-a8f7fd77081d. However, the model in this article is still heavy.