NVIDIA-AI-IOT/trt_pose #86: Models with higher input size

lweicker, 2020-12-02:
I'm really impressed with the results obtained so far.
I'm wondering if you plan to train and publish models but with higher input size? 
I ask that because I see in `tasks/human_pose/experiments/` that you experimented on higher input image sizes like 368x368, 384x384.

Azzedine-Touazi, 2020-12-09:
Plz, how can we use TRT-pose for high resolution data (e.g. 1920 x 1080 images)?

Thanks in advance.

jaybdub, 2020-12-09:
Hi all,

We did a few experiments with higher resolutions, but for our primary use case (pose detection within a few meters using an Raspberry pi camera), we didn't find much qualitative improvements, and the runtime increased.  If you want to detect objects within a few meters using a typical camera (say 50+degree FOV), I'd recommend just downscaling the image before providing to the neural network.

That said, you *can* run the existing pre-trained models at higher resolution, but it will change the effective size range of objects you detect.  To do this, you need to adjust the size of the input data you provide to the model when optimizing with TensorRT.  This has not been thoroughly tested, so your results may vary.

Please let me know if this helps or you have further questions.

Best,
John

lweicker, 2020-12-11:
Hi John,

Thanks for your answer. The inference with higher input size indeed works as you explained. After running multiple tests, I however have the feeling that the quality of the predictions is not as good as resizing first to either 224*224 or 256*256 (depending on the model used) and then infer.

It also led me to an other question. During the training, how did you resize the images? Did you squish them or cropped them?

Best,

Lionel

Azzedine-Touazi, 2021-01-21:
Hello John,

I tested the code with 1280x720 camera resolution and 15 FPS by modifying live_demo.ipynb. This was performed by setting the two variables WIDTH and HEIGHT to 1280 and 780, respectively and using 224x224 model. 

I have observed that the prediction results are not always satisfactory with excessive runtime (the estimated throughput in FPS is 2.5 with Jetson TX2) . 

In certain situations (when the subject is near to the camera), there are some keypoints detected completely outside the human body. 

I guess if I resize the images, before inference, the prediction quality/runtime will be improved. However, which technique/strategy would you recommend for resizing images? Is there any other parameter that need to be adjusted? Please advise.

Thanks.

Azzedine.