NVIDIA-AI-IOT/trt_pose #91: Increase FPS

mcdonasd, 2020-12-07:
I am using trt_pose  on a system with i5 cpu and GTX 1660 Super GPU.   I am currently only getting 9 fps for an 1080 x 1920 video.  Any ideas for how to improve FPS?

jaybdub, 2020-12-09:
Hi mcdonasd,

Thanks for reaching out!

Are you downscaling the image before providing it to the neural network?  I would recommend downscaling to 224x224 or 256x256 if you are trying to detect people within a few meters.  The output can be re-scaled to match your original image dimensions.

Higher resolutions may work, but will change the effective size of objects you detect.  The higher resolution the image, the smaller the size of object you will detect, and slower the model will run.  It is best to tune the input shape for your application.  For example, a human-computer interaction robot probably only needs 224x224, monitoring pose of a crowd at a distance, something much higher.

Please let me know if this helps, or you run into any issues.

Best,
John

mcdonasd, 2020-12-10:
Thank you, I will implement your recommendations and keep you posted.  Since I posted, I realized implementing the recommendations in Issue #79 slowed video playback significantly.  I am still working thru how to save the keypoints to a dataframe while minimizing the performance impact.   I would welcomed ideas / recommendations.

Thank you!

mcdonasd, 2020-12-10:
@jaybdub  I have confirmed that I am downscaling to 224x224.  In this application, we need to only detect the largest person in the image.

mcdonasd, 2020-12-19:
Speed was significantly improved when I used a dictionary instead of dataframe to remap the joints.

closed 2020-12-19 by mcdonasd

tucachmo2202, 2021-03-10:
> Speed was significantly improved when I used a dictionary instead of dataframe to remap the joints.

Hi @mcdonasd,
Could you share your code using dictionary? Tks.