NVIDIA-AI-IOT/trt_pose #97: RuntimeError: NVRTC error: while running cell-5

arvindchandel, 2020-12-16:
While running cell-5 of live_demo.ipynb notebook
import torch2trt
model_trt = torch2trt.torch2trt(model, [data], fp16_mode=False, max_workspace_size=1<<25)  
getting error below:
RuntimeError                              Traceback (most recent call last)
<ipython-input-6-3e4014f687f8> in <module>
      1 import torch2trt
      2 
----> 3 model_trt = torch2trt.torch2trt(model, [data], fp16_mode=False, max_workspace_size=1<<25)

~/tlt-experiments/skeleton-env/lib/python3.6/site-packages/torch2trt-0.1.0-py3.6-linux-x86_64.egg/torch2trt/torch2trt.py in torch2trt(module, inputs, input_names, output_names, log_level, max_batch_size, fp16_mode, max_workspace_size, strict_type_constraints, keep_network, int8_mode, int8_calib_dataset, int8_calib_algorithm, int8_calib_batch_size, use_onnx)
    553         )
    554 
--> 555     engine = builder.build_cuda_engine(network)
    556 
    557     module_trt = TRTModule(engine, input_names, output_names)

RuntimeError: NVRTC error:

arvindchandel, 2020-12-17:
Issue sorted

closed 2020-12-17 by arvindchandel

igormusinov, 2021-04-08:
Hi @arvindchandel 

How did you solve this issue?

arvindchandel, 2021-04-08:
@igormusinov tried 4 month ago, don't remember exactly.