NVIDIA-AI-IOT/trt_pose #98: Only select the person closest to the camera

mcdonasd, 2020-12-18:
Has anyone modified this solution to only detect the person closest to the camera?

mcdonasd, 2020-12-19:
Is the solution as easy as the following?
```
counts, objects, peaks = parse_objects(cmap, paf)

if int(counts) == 1:
      draw_objects(image, counts, objects, peaks)
```

cwlinghk, 2021-01-07:
> Is the solution as easy as the following?
> 
> ```
> counts, objects, peaks = parse_objects(cmap, paf)
> 
> if int(counts) == 1:
>       draw_objects(image, counts, objects, peaks)
> ```
counts is the number of detected people.
Your case would just draw the skeleton if there is one people detected.
I guess you have to calculate the area of the bounding box, find maximum, and select the people to draw

mcdonasd, 2021-01-07:
Thank you!  This helps.

closed 2021-01-07 by mcdonasd

djramakrishna, 2021-02-19:
Hey @mcdonasd  I'm looking for something very similar. I want to make sure it detects only one person per frame (which has teh highest confidence score) and return only one value for each key-point in that particular frame. Please let me know if you've solved this issue.