NVIDIA-AI-IOT/trt_pose #100: Cuda

dexmac221, 2020-12-19:
Hi All
I successfully installed trt_pose on jetson nano board, very interesting, thanks! i would like to install on a pc with 2070rtx using  nvcr.io/nvidia/tensorrt:20.12-py3 based custom docker but, during model conversion (early), it gives me this error

[TensorRT] ERROR: CUDA initialization failure with error 222. Please check your CUDA installation:  http://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html
Traceback (most recent call last):
  File "live_demo.py", line 40, in <module>
    model_trt = torch2trt.torch2trt(model, [data], fp16_mode=True, max_workspace_size=1<<25)
  File "/usr/local/lib/python3.8/dist-packages/torch2trt-0.1.0-py3.8.egg/torch2trt/torch2trt.py", line 498, in torch2trt
TypeError: pybind11::init(): factory function returned nullptr

ubuntu            18.04
Driver Version: 450.80.02
cuda               11.1
torch               1.7.1
torch2trt          0.1.0
torchvision      0.8.2

Thanks in advance
Gianluca

dexmac221, 2020-12-20:
Hi All
Solved with nvcr.io/nvidia/tensorrt:19.12-py3 docker image

Thanks
Gianluca

closed 2020-12-20 by dexmac221

gioipv, 2021-09-11:
```
[TensorRT] ERROR: CUDA initialization failure with error 222. Please check your CUDA installation:  http://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html
Traceback (most recent call last):
  File "tensorrt_convert2trt.py", line 54, in <module>
    main(args)
  File "tensorrt_convert2trt.py", line 35, in main
    model_trt = torch2trt(model, [input_], max_workspace_size=1 << 30)
  File "/home/gioipv/workspaces/miniconda3/envs/eye_state/lib/python3.7/site-packages/torch2trt-0.3.0-py3.7.egg/torch2trt/torch2trt.py", line 519, in torch2trt
    builder = trt.Builder(logger)
TypeError: pybind11::init(): factory function returned nullptr
```

Helllo, I am newbie
In my case, i think my user was not in docker group for user docker on nvidia.
I use aws and GPU is Tesla T4. Tensorrt use docker on nvidia.
Because, after I added my user account into [docker group](https://docs.docker.com/engine/install/linux-postinstall/) with  `sudo usermod -aG docker $USER` 
I have to  activate the changes to groups with `newgrp docker` every time to use Tensorrt. Maybe after reboot my aws, i dont have to command `newgrp docker`, but for now, I had to do it for fix this issues