NVIDIA-AI-IOT/trt_pose #101: import-im6.q16: not authorized `re'

bobby20180331, 2020-12-21:
errors occur when convert .pth to onnx . I just follow the deep_stream_trt_pose instrution 
use :
./export_for_isaac.py --input_checkpoint resnet18_baseline_att_224x224_A_epoch_249.pth
erros:
: File name too long
import-im6.q16: not authorized `re' @ error/constitute.c/WriteImage/1037.
import-im6.q16: not authorized `torch' @ error/constitute.c/WriteImage/1037.
./export_for_isaac.py: line 60: syntax error near unexpected token `('
./export_for_isaac.py: line 60: `class InputReNormalization(torch.nn.Module):'

kaisark, 2021-05-05:
@bobby20180331 Try the following:

username@nano: python3 ./export_for_isaac.py --input_checkpoint resnet18_baseline_att_224x224_A_epoch_249.pth --input_width 224 --input_height 224 --output_model pose_estimation.onnx  --input_topology /home/username/trt_pose/tasks/human_pose/human_pose.json 

Input model is not specified, using resnet18_baseline_att as a default.
Successfully completed convertion of resnet18_baseline_att_224x224_A_epoch_249.pth to pose_estimation.onnx.

bobby20180331, 2021-05-08:
thanks.  it worked!

closed 2021-05-08 by bobby20180331