NVIDIA-AI-IOT/trt_pose #111: OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

perng, 2021-02-22:
Hi, I'm encountering this problem when running live_demo.ipynb
Please help

--------------------------------------------------------
OSError                Traceback (most recent call last)
<ipython-input-1-0c2219f9314e> in <module>
      1 import json
----> 2 import trt_pose.coco
      3 
      4 with open('human_pose.json', 'r') as f:
      5     human_pose = json.load(f)

/usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-aarch64.egg/trt_pose/__init__.py in <module>
----> 1 import torch

~/.local/lib/python3.6/site-packages/torch/__init__.py in <module>
    187     # See Note [Global dependencies]
    188     if USE_GLOBAL_DEPS:
--> 189         _load_global_deps()
    190     from torch._C import *
    191 

~/.local/lib/python3.6/site-packages/torch/__init__.py in _load_global_deps()
    140     lib_path = os.path.join(os.path.dirname(here), 'lib', lib_name)
    141 
--> 142     ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
    143 
    144 

/usr/lib/python3.6/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    346 
    347         if handle is None:
--> 348             self._handle = _dlopen(self._name, mode)
    349         else:
    350             self._handle = handle

OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

Anqi-nus, 2021-03-04:
same here, have you solved the problem?

perng, 2021-03-04:
yes, I solved it by

export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1

btw, folk on Nvidia forum is more responsive. I got question answered there. 
https://forums.developer.nvidia.com/t/what-is-cannot-allocate-memory-in-static-tls-block/169225

wuhaoqi-1, 2021-04-06:
> yes, I solved it by
> 
> export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1
> 
> btw, folk on Nvidia forum is more responsive. I got question answered there.
> https://forums.developer.nvidia.com/t/what-is-cannot-allocate-memory-in-static-tls-block/169225

hello,this problem still occurs after I follow this command to run
[image]

perng, 2021-04-06:
I don't have solution for you. I suggest you go to the Nvidia forum to get help. 

> hello,this problem still occurs after I follow this command to run
> [image]

AlexandreBourrieau, 2021-08-30:
Hello,

Got the same problem and this solution not work for me