NVIDIA-AI-IOT/trt_pose #112: The problem of No module named 'trt_pose.plugins'

yantaixu0120, 2021-02-25:
When I run the live_demo_v2.ipynb script in the jupyter notebook,I encountered the follow problem,Can somebody help me,thanks!!!

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-3-7c97b68d0a67> in <module>
      1 import json
----> 2 import trt_pose.coco
      3 
      4 with open('human_pose.json', 'r') as f:
      5     human_pose = json.load(f)

~/WuHanGangTie/WG_pose/tasks/human_pose/trt_pose/coco.py in <module>
      7 import tqdm
      8 import trt_pose
----> 9 import trt_pose.plugins
     10 import glob
     11 import torchvision.transforms.functional as FT

ModuleNotFoundError: No module named 'trt_pose.plugins'

Hiroaki-K4, 2021-04-22:
@yantaixu0120 
Hi, Have you solved?
I have a same problem.

yantaixu0120, 2021-04-25:
> @yantaixu0120
> Hi, Have you solved?
> I have a same problem.

Hi,You can follow the solution in this link..
https://github.com/NVIDIA-AI-IOT/trt_pose/issues/2 
after building, please put plugins.cpython-36m-x86_64-linux-gnu.so who is under buid/lib.linux-x86_64-3.6/trt_pose/ into /trt_pose/. should be fine after that.