NVIDIA-AI-IOT/trt_pose #113: Erros about my own model which deteced only three keypoingts when lodading models

yantaixu0120, 2021-02-26:
I have trained my own models use the resnet50 network and config files "resnet50_baseline_att_384x384_A.json", 
when I want do the inference using the following code :

“”“”“”“”“”“”“”“
import json
import trt_pose.coco
with open('dmcode.json', 'r') as f:
    dmcode_pose = json.load(f)
topology = trt_pose.coco.coco_category_to_topology(dmcode_pose)

import trt_pose.models
num_parts = len(dmcode_pose['keypoints'])
print(num_parts)
num_links = len(dmcode_pose['skeleton'])
print(num_links)
model = trt_pose.models.resnet50_baseline_att(num_parts, 2 * num_links).cuda().eval()

import torch
#MODEL_WEIGHTS = 'resnet18_baseline_att_224x224_A_epoch_249.pth'
MODEL_WEIGHTS = 'resnet50_baseline_att_384x384_A_epoch_50.pth'
model.load_state_dict(torch.load(MODEL_WEIGHTS))
”“”“”“”“”“”“”“”“”“”“”“”“

when loading models I encountered erros as below:

---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-12-c08518b974e4> in <module>
      4 MODEL_WEIGHTS = 'resnet50_baseline_att_384x384_A_epoch_50.pth'
      5 
----> 6 model.load_state_dict(torch.load(MODEL_WEIGHTS))

~/anaconda3/envs/wg_detect_yolov5/lib/python3.8/site-packages/torch/nn/modules/module.py in load_state_dict(self, state_dict, strict)
   1049 
   1050         if len(error_msgs) > 0:
-> 1051             raise RuntimeError('Error(s) in loading state_dict for {}:\n\t{}'.format(
   1052                                self.__class__.__name__, "\n\t".join(error_msgs)))
   1053         return _IncompatibleKeys(missing_keys, unexpected_keys)

RuntimeError: Error(s) in loading state_dict for Sequential:
	Missing key(s) in state_dict: "1.cmap_up.6.weight", "1.cmap_up.6.bias", "1.cmap_up.7.weight", "1.cmap_up.7.bias", "1.cmap_up.7.running_mean", "1.cmap_up.7.running_var", "1.paf_up.6.weight", "1.paf_up.6.bias", "1.paf_up.7.weight", "1.paf_up.7.bias", "1.paf_up.7.running_mean", "1.paf_up.7.running_var". 

What should I do to solve this problem??

lian-yeh, 2021-08-17:
I have same problem.
please can help me to solve the problem

thank you

nanmi, 2021-12-15:
```python
import json
import trt_pose.coco
import trt_pose.models
import torch

class InputReNormalization(torch.nn.Module):
    """
        This defines "(input - [0.485, 0.456, 0.406]) / [0.229, 0.224, 0.225]" custom operation
        to conform to "Unit" normalized input RGB data.
    """
    def __init__(self):
        super(InputReNormalization, self).__init__()
        self.mean = torch.Tensor([0.485, 0.456, 0.406]).reshape((1,3,1,1)).cuda()
        self.std = torch.Tensor([0.229, 0.224, 0.225]).reshape((1,3,1,1)).cuda()

    def forward(self, x):
        return (x - self.mean) / self.std

class HeatmapMaxpoolAndPermute(torch.nn.Module):
    """
        This defines MaxPool2d(kernel_size = 3, stride = 1) and permute([0,2,3,1]) custom operation
        to conform to [part_affinity_fields, heatmap, maxpool_heatmap] output format.
    """
    def __init__(self):
        super(HeatmapMaxpoolAndPermute, self).__init__()
        self.maxpool = torch.nn.MaxPool2d(3, stride=1, padding=1)

    def forward(self, x):
        heatmap, part_affinity_fields = x
        maxpool_heatmap = self.maxpool(heatmap)

        part_affinity_fields = part_affinity_fields.permute([0,2,3,1])
        heatmap = heatmap.permute([0,2,3,1])
        maxpool_heatmap = maxpool_heatmap.permute([0,2,3,1])
        return [part_affinity_fields, heatmap, maxpool_heatmap]

HUMAN_POSE = 'human_pose.json'
MODEL_WEIGHTS = 'resnet18_baseline_att_224x224_A_epoch_249.pth'
WIDTH = 640
HEIGHT = 640

with open(HUMAN_POSE, 'r') as f:
    human_pose = json.load(f)

topology = trt_pose.coco.coco_category_to_topology(human_pose)
num_parts = len(human_pose['keypoints'])
num_links = len(human_pose['skeleton'])
model = trt_pose.models.resnet18_baseline_att(num_parts, 2 * num_links).cuda().eval()
model.load_state_dict(torch.load(MODEL_WEIGHTS))

# Add InputReNormalization pre-processing and HeatmapMaxpoolAndPermute post-processing operations
model = torch.nn.Sequential(InputReNormalization(), model, HeatmapMaxpoolAndPermute())

# Define input and output names for ONNX exported model.
input_names = ["input"]
output_names = ["part_affinity_fields", "heatmap", "maxpool_heatmap"]
dummy_input = torch.zeros((1, 3, HEIGHT, WIDTH)).cuda()
torch.onnx.export(model, dummy_input, "resnet18_baseline_att_224x224_A_epoch_249.onnx",
                    input_names=input_names, output_names=output_names, opset_version=11)

```