NVIDIA-AI-IOT/trt_pose #115: Low FPS than report in Jetson nano

tucachmo2202, 2021-03-03:
Hi, 
I highly appreciate your repo. But I test your model in Jetson nano Tegra X1 it just got about 7FPS lower than 22 FPS in your report. I just follow your guide except I did not use USB camera, I used OpenCV to read image instead of. I want to know whether you use other kind of Jetson nano or I made a mistake. Many tks.

twmht, 2021-03-26:
I have the same issue.

tucachmo2202, 2021-03-26:
@twmht I guess the author calculate with small cmap_window and quite small max_num_objects.

alessandrolombardini, 2021-04-07:
I have the same issue, I think. With resnet I reached only 6 FPS. I used the notebook given so it's quite strange that I can't get the same results of authors using the same environment.

closed 2021-08-30 by tucachmo2202