NVIDIA-AI-IOT/trt_pose #118: Increase resolution

jptalledo10, 2021-03-09:
Hi

Can you advise the steps in order to handle a  resolution up to 1080 if possible? Which configuration settings should be changed? is re-training required?

Trying to achieve pose detection but at long distance.

Thanks

jaybdub, 2021-03-19:
Hi jptalledo10,

Thanks for reaching out!

If you expect all objects will be of a similar scale, It may be worth trying the model directly on a larger image to figure out which scale works for your application.

At first, I would suggest doing this with the pre-trained model, simply changing the size of the image when you optimize with TensorRT.

If the scale of the objects varies greatly (like full image size - small block of pixels), it is unlikely the models would handle all scales out of the box.  You can consider they are good at detecting objects with a size of about 112-224 pixels tall (I'm assuming for the 224px image they detect objects down to about half the image height).

Please let me know if this helps or you have any questions.

Best,
John

jptalledo10, 2021-03-21:
@jaybdub 

I assume you are suggesting to change the following:

WIDTH = 224
HEIGHT = 224
data = torch.zeros((1, 3, HEIGHT, WIDTH)).cuda()

To a bigger image size from the camera source? like 1920x1080? 

Can you elaborate further about the pre-trained model? 

Thanks

troymyname, 2024-04-26:
Hi @jaybdub, can you elaborate more on how to use larger size images?