NVIDIA-AI-IOT/trt_pose #129: model not working on multi person

aniketzz, 2021-05-26:
When I run the model on two-person in 1 frame, it's working fine. But when more than 2 person comes in the frame, the 2nd person's keypoints gets distributed between the 2nd and 3rd or nth person. 

I am unable to pinpoint the issue. The model is not able to decide which coordinate belongs to which person in the case of more than 2 people.

I have attached the screenshots of the results for referance.
[image]
[image]
[image]

nireshd, 2021-07-20:
Yeah, i have the same problem. model only detects two people

HZNUJeffreyRen, 2021-09-30:
I have the same problem, Did you solve it？

aniketzz, 2021-09-30:
I did the same thing in [Deepstream](https://developer.nvidia.com/blog/creating-a-human-pose-estimation-application-with-deepstream-sdk/).  Current it is set to detect only 2 people by default. Changing `max_num_objects = 2` to max_num_objects = 100` did the trick.