NVIDIA-AI-IOT/trt_pose #135: How to get x, y coordinates of neck?

Amyhds, 2021-06-22:
I want to see the changes of coordinates of neck. 
How can I get these?

pramod-wick, 2021-06-22:
Hi,

you can follow the code given in this blog of using trt-pose (works on pc or jetson device). Specially use the "get_keypoint" function to extract keypoints of any joint.
 
[https://spyjetson.blogspot.com/2019/12/jetsonnano-human-pose-estimation-using.html](url)

Amyhds, 2021-06-22:
> Hi,
> 
> you can follow the code given in this blog of using trt-pose (works on pc or jetson device). Specifically use the "get_keypoint" function to extract keypoints of any joint.
> 
> https://spyjetson.blogspot.com/2019/12/jetsonnano-human-pose-estimation-using.html

Thank you so much

Amyhds, 2021-07-04:
There's something wrong with opencv....I got some errors...

closed 2021-11-30 by Amyhds