NVIDIA-AI-IOT/trt_pose #137: ModuleNotFoundError: No module named 'torch'

khalidh223, 2021-06-22:
I am trying to run this line from `README.md`: `sudo python3 setup.py install --plugins` in part 2 of it, but I keep getting the error ```ModuleNotFoundError: No module named 'torch'```.  But I have already installed torch, torchvision, and torchaudio with ```pip3 install torch==1.9.0+cpu torchvision==0.10.0+cpu torchaudio==0.9.0 -f https://download.pytorch.org/whl/torch_stable.html```, so I do not know why I am still seeing this error. I am running Ubuntu 20.4 on Virtual Machine, and have Python 3.8.5 installed on it.

khalidh223, 2021-06-23:
I found out that I had not installed torch, torchvision, and torchaudio on sudo, so I just did. Ran the same line again for setup.py, and got the error ```OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.``` Do I have to install CUDA do utilize this?

korcsogdenes, 2021-07-22:
Probably, since this uses tensorRT which uses CUDA. Maybe it's possible to move it to the cpu, but you will lose the optimizations.