NVIDIA-AI-IOT/trt_pose #154: Problem with live_demo.ipynb

AK51, 2022-01-15:
Hi,
When I try to run the demo, the first line has error.
http://localhost:8888/lab/tree/trt_pose/tasks/human_pose
May I know which part I maybe missing? I am using nano.

Thanks,

'''
import json
import trt_pose.coco
​
with open('human_pose.json', 'r') as f:
    human_pose = json.load(f)
​
topology = trt_pose.coco.coco_category_to_topology(human_pose)
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-3-0c2219f9314e> in <module>
      1 import json
----> 2 import trt_pose.coco
      3 
      4 with open('human_pose.json', 'r') as f:
      5     human_pose = json.load(f)

/usr/local/lib/python3.6/dist-packages/trt_pose-0.0.1-py3.6-linux-aarch64.egg/trt_pose/__init__.py in <module>
----> 1 import torch

~/.local/lib/python3.6/site-packages/torch/__init__.py in <module>
    186     # See Note [Global dependencies]
    187     if USE_GLOBAL_DEPS:
--> 188         _load_global_deps()
    189     from torch._C import *
    190 

~/.local/lib/python3.6/site-packages/torch/__init__.py in _load_global_deps()
    139     lib_path = os.path.join(os.path.dirname(here), 'lib', lib_name)
    140 
--> 141     ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
    142 
    143 

/usr/lib/python3.6/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    346 
    347         if handle is None:
--> 348             self._handle = _dlopen(self._name, mode)
    349         else:
    350             self._handle = handle

OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block
'''

mingun0112, 2022-05-25:
same problem with U....

mingun0112, 2022-07-27:
> me too…

I solved this problem few months ago
so, I can't remember how to solve clearly.
I think you should add some path to ~/.profile
I find out later I don't need to use live demo because I just need onnx file!