NVIDIA-AI-IOT/trt_pose #180: Unable to Access Pre-trained Models

K-Wince, 2024-10-03:
When I click on the link to download the model weights, it shows this page:
[image]

How would I go about getting access to those files to run trt_pose?

tdes-ut, 2024-11-13:
Same here. It seems like an inappropriate (possibly inadvertent?) choice to host it in such a way that access is restricted and needs to be manually approved. It seems like development on this repo is not active, though, so not sure anyone will come through for us.

@fzoric8 wanna share with the class?

TinkerSo, 2024-11-18:
> Same here!
> 
> P.S. @K-Wince contact me, I may have found a solution.

HI, I was wondering how you were able to download the weights, also seeking a way around the permission access problem. Thank you in advance!

fzoric8, 2024-11-19:
Hi @TinkerSo and @tdes-ut , 

You can find weights [here](https://github.com/make2explore/Real-Time-Hand-Pose-Estimation-on-Jetson-Nano/tree/main/Pre-Trained%20Models), as I later found out. 

Just beware because folder names are swapped: `trt_pose` contains weights for `trt_hand_pose`, and `trt_hand_pose` contains wieghts for `trt_pose`. 

Best, 
Filip

itsme-jung, 2024-11-21:
@fzoric8 
hello! I want to ask that ...
so it works on trt_hand_pose link you mentioned?? without any accept or authority issue? 

then do we just need setups mentioned in --> https://github.com/NVIDIA-AI-IOT/trt_pose/blob/master/README.md
including step1 and step2? 
thank you so much,,

Dasha0712, 2025-02-20:
thanks it works